suculent/esp32-http-update #1: Compilation error

bhaessig83, 2018-02-13:
Thank you for the library, but I get the following error while trying to compile.

I've updated the ESP32 library but it has no effect.

Thank you for your help

`
C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp: In member function 'virtual std::unique_ptr<WiFiClient> TLSTraits::create()':

C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp:64:66: error: no matching function for call to 'std::unique_ptr<WiFiClient>::unique_ptr(WiFiClientSecure*)'

         return std::unique_ptr<WiFiClient>(new WiFiClientSecure());

                                                                  ^

In file included from c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\memory:81:0,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiClient.h:28,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFi.h:37,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp:29:

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:228:2: note: candidate: template<class _Up, class> std::unique_ptr<_Tp, _Dp>::unique_ptr(std::auto_ptr<_Up>&&)

  unique_ptr(auto_ptr<_Up>&& __u) noexcept;

  ^

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:228:2: note:   template argument deduction/substitution failed:

C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp:64:66: note:   mismatched types 'std::auto_ptr<_Up>' and 'WiFiClientSecure*'

         return std::unique_ptr<WiFiClient>(new WiFiClientSecure());

                                                                  ^

In file included from c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\memory:81:0,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiClient.h:28,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFi.h:37,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp:29:

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:220:2: note: candidate: template<class _Up, class _Ep, class> std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Up, _Ep>&&)

  unique_ptr(unique_ptr<_Up, _Ep>&& __u) noexcept

  ^

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:220:2: note:   template argument deduction/substitution failed:

C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp:64:66: note:   mismatched types 'std::unique_ptr<_Tp, _Dp>' and 'WiFiClientSecure*'

         return std::unique_ptr<WiFiClient>(new WiFiClientSecure());

                                                                  ^

In file included from c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\memory:81:0,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFiClient.h:28,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/WiFi.h:37,

                 from C:\Users\bhaessig\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src\HTTPClient.cpp:29:

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:205:7: note: candidate: std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Tp, _Dp>&&) [with _Tp = WiFiClient; _Dp = std::default_delete<WiFiClient>]

       unique_ptr(unique_ptr&& __u) noexcept

       ^

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:205:7: note:   no known conversion for argument 1 from 'WiFiClientSecure*' to 'std::unique_ptr<WiFiClient>&&'

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:200:17: note: candidate: constexpr std::unique_ptr<_Tp, _Dp>::unique_ptr(std::nullptr_t) [with _Tp = WiFiClient; _Dp = std::default_delete<WiFiClient>; std::nullptr_t = std::nullptr_t]

       constexpr unique_ptr(nullptr_t) noexcept : unique_ptr() { }

                 ^

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:200:17: note:   no known conversion for argument 1 from 'WiFiClientSecure*' to 'std::nullptr_t'

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:193:7: note: candidate: std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Tp, _Dp>::pointer, typename std::remove_reference<_To>::type&&) [with _Tp = WiFiClient; _Dp = std::default_delete<WiFiClient>; std::unique_ptr<_Tp, _Dp>::pointer = WiFiClient*; typename std::remove_reference<_To>::type = std::default_delete<WiFiClient>]

       unique_ptr(pointer __p,

       ^

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:193:7: note:   candidate expects 2 arguments, 1 provided

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:181:7: note: candidate: std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Tp, _Dp>::pointer, typename std::conditional<std::is_reference<_Dp>::value, _Dp, const _Dp&>::type) [with _Tp = WiFiClient; _Dp = std::default_delete<WiFiClient>; std::unique_ptr<_Tp, _Dp>::pointer = WiFiClient*; typename std::conditional<std::is_reference<_Dp>::value, _Dp, const _Dp&>::type = const std::default_delete<WiFiClient>&]

       unique_ptr(pointer __p,

       ^

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:181:7: note:   candidate expects 2 arguments, 1 provided

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:169:7: note: candidate: std::unique_ptr<_Tp, _Dp>::unique_ptr(std::unique_ptr<_Tp, _Dp>::pointer) [with _Tp = WiFiClient; _Dp = std::default_delete<WiFiClient>; std::unique_ptr<_Tp, _Dp>::pointer = WiFiClient*]

       unique_ptr(pointer __p) noexcept

       ^

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:169:7: note:   no known conversion for argument 1 from 'WiFiClientSecure*' to 'std::unique_ptr<WiFiClient>::pointer {aka WiFiClient*}'

c:\users\bhaessig\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits\unique_ptr.h:157:17: note: candidate: constexpr std::unique_ptr<_Tp, _Dp>::unique_ptr() [with _Tp = WiFiClient; _Dp = std::default_delete<WiFiClient>]

       constexpr unique_ptr() noexcept

                 ^
`

suculent, 2018-02-16:
I've updated to latest Arduino ESP32 Core but cannot reproduce this. Seems like mismatch between Arduino and ESP32 HTTPClient libs.

My compilation ends up with those:
```
Multiple libraries were found for "WiFi.h"
 Used: .../Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Using library WiFi at version 1.0 in folder: .../Documents/Arduino/hardware/espressif/esp32/libraries/WiFi 
Using library HTTPClient at version 1.1 in folder: .../Documents/Arduino/hardware/espressif/esp32/libraries/HTTPClient 
Using library WiFiClientSecure at version 1.0 in folder: .../Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure 
Using library ESP32httpUpdate at version 2.1.145 in folder: .../Documents/Arduino/libraries/ESP32httpUpdate 
Using library Update at version 1.0 in folder: .../Documents/Arduino/hardware/espressif/esp32/libraries/Update 
Using library FS at version 1.0 in folder: .../Documents/Arduino/hardware/espressif/esp32/libraries/FS 
Using library SPIFFS at version 1.0 in folder: .../Documents/Arduino/hardware/espressif/esp32/libraries/SPIFFS 
```

bhaessig83, 2018-02-19:
I've tested with an older release of the esp32 library and the compilation is working. I've saved now this version in case of new updates.
Thank you for your help

suculent, 2018-02-19:
What was the version you had issue with?
Is this still an issue or should I close it?

Thanks. 

Odesláno z iPhonu

bhaessig83, 2018-02-19:
The compilation was working with this old library version : https://github.com/espressif/arduino-esp32/tree/a38ffe58fe1d73a56295edb0c6051a46932e0175
But still not working with the current one.

On my side, we can close it, but maybe other people will have the same problem and the issue has to be solved in the current library.

suculent, 2018-02-19:
It seems to build just fine here against:
```
$ git rev-list head --max-count=1
a153f6ad280e5932be9c80341556348ff505d586
```

suculent, 2020-05-11:
See TL;DR in README, this is already deprecated and implemented in Arduino Core.

closed 2020-05-11 by suculent