suculent/esp32-http-update #2: SPIFFS mounting issue

basic-user, 2018-02-28:
Hi suculent,
thank you so much for the library this is very useful! The firmware OTA works great! I am trying to do a SPIFFS OTA and I face some issues.
I am using the SPIFFS_update.ino example and it seems to be working fine: it find the file, download it and write it.
But when I try to mount a SPIFFS file system to look and read the file that was written it looks like nothing was written.

```
#include <Arduino.h>

#include <WiFi.h>

#include <HTTPClient.h>
#include <ESP32httpUpdate.h>
#include "FS.h"
#include "SPIFFS.h"
#define USE_SERIAL Serial

void listDir(fs::FS &fs, const char * dirname, uint8_t levels) {
	Serial.printf("Listing directory: %s\n", dirname);

	File root = fs.open(dirname);
	if (!root) {
		Serial.println("Failed to open directory");
		return;
	}
	if (!root.isDirectory()) {
		Serial.println("Not a directory");
		return;
	}

	File file = root.openNextFile();
	while (file) {
		if (file.isDirectory()) {
			Serial.print("  DIR : ");
			Serial.println(file.name());
			if (levels) {
				listDir(fs, file.name(), levels - 1);
			}
		}
		else {
			Serial.print("  FILE: ");
			Serial.print(file.name());
			Serial.print("  SIZE: ");
			Serial.println(file.size());
		}
		file = root.openNextFile();
	}
}

void setup() {

	USE_SERIAL.begin(115200);
	USE_SERIAL.setDebugOutput(true);

	USE_SERIAL.println();
	USE_SERIAL.println();
	USE_SERIAL.println();
	WiFi.begin("SSID", "password");
	/*for (uint8_t t = 4; t > 0; t--) {
		USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
		USE_SERIAL.flush();
		delay(1000);
	}*/

	const esp_partition_t* partition1 = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, NULL);

	Serial.print("Partition type: ");
	Serial.println(partition1->type);
	Serial.print("Partition subtype: ");
	Serial.println(partition1->subtype);
	Serial.print("Partition size: ");
	Serial.println(partition1->size);
	Serial.print("Partition address: ");
	Serial.println(partition1->address);
	Serial.print("Partition label: ");
	Serial.println(partition1->label);
	Serial.print("Partition encrypted: ");
	Serial.println(partition1->encrypted);

	SPIFFS.begin(true);
	Serial.println("Used Bytes: " + String(SPIFFS.usedBytes()));
	listDir(SPIFFS, "/", 0);
	SPIFFS.end();
	//SPIFFS.format();
}

void loop() {
	// wait for WiFi connection
	if ((WiFi.status() == WL_CONNECTED)) {

		USE_SERIAL.println("Update SPIFFS...");
		t_httpUpdate_return ret = ESPhttpUpdate.updateSpiffs("http://binaries.climate-solutions.net/update_TFT.php", "1.2.9");
		switch (ret) {
		case HTTP_UPDATE_FAILED:
			USE_SERIAL.printf("HTTP_UPDATE_FAILED Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
			break;

		case HTTP_UPDATE_NO_UPDATES:
			USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
			break;

		case HTTP_UPDATE_OK:
			USE_SERIAL.println("HTTP_UPDATE_OK");
			break;
		}
		if (ret == HTTP_UPDATE_OK) {

			
		}
		SPIFFS.begin(false);
		listDir(SPIFFS, "/", 0);
		Serial.println("Used Bytes: " + String(SPIFFS.usedBytes()));
	}
	delay(20000);
}
```
I added debug messages in your library and the Updater.cpp file to try to understand what is happening.
Here is the debug I get form the serial port:

> Opening port
> Port open
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371 
> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:956
> load:0x40078000,len:0
> load:0x40078000,len:13076
> entry 0x40078ad0
> 
> 
> 
> Partition type: 1
> Partition subtype: 130
> Partition size: 1503232
> Partition address: 2691072
> Partition label: spiffs
> Partition encrypted: [D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
> 0
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)
> Tasks currently running:
> CPU 0: ipc0
> CPU 1: IDLE
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)
> Tasks currently running:
> CPU 0: ipc0
> CPU 1: IDLE
> Used Bytes: 0
> Listing directory: /
> [D][WiFiGeneric.cpp:265] _eventCallback(): Event: 4 - STA_CONNECTED
> [D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP
> Update SPIFFS...
> [V][HTTPClient.cpp:140] beginInternal(): url: http://random-server/update_TFT.php
> [D][HTTPClient.cpp:181] beginInternal(): host: random-server port: 80 url: /update_TFT.php
> [D][HTTPClient.cpp:850] connect():  connected to random-server:80
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'HTTP/1.0 200 OK'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'X-Powered-By: PHP/7.0.27'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Type: application/octet-stream'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Disposition: attachment; filename=Test_TFT.txt'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'x-MD5: f60ce1060c117036ec9aabaa45c3b129'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Length: 186128'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Date: Wed, 28 Feb 2018 14:50:33 GMT'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Accept-Ranges: bytes'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Server: LiteSpeed'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Connection: close'
> [V][HTTPClient.cpp:938] handleHeaderResponse(): RX: ''
> [D][HTTPClient.cpp:968] handleHeaderResponse(): code: 200
> [D][HTTPClient.cpp:971] handleHeaderResponse(): size: 186128
> Starting SPIFFS Update
> Starting tcp
> runUpdate spiffs...
> begin update
> Partition type: 1
> Partition subtype: 130
> Partition size: 1503232
> Partition address: 2691072
> Partition label: spiffs
> Partition encrypted: 0
> check md5
> writting stream
> ending update
> [V][HTTPClient.cpp:244] end(): tcp is closed
> [V][HTTPClient.cpp:244] end(): tcp is closed
> HTTP_UPDATE_OK
> E (36611) SPIFFS: mount failed, -10025
> [E][SPIFFS.cpp:47] begin(): Mounting SPIFFS failed! Error: -1
> Listing directory: /
> [E][vfs_api.cpp:22] open(): File system is not mounted
> Failed to open directory
> Used Bytes: 0

It looks like I can not mount the SPIFFS without reformatting. I tried to mount it before the update and leave it mounted, but it seems nothing is written. And I tried to un-mount it before doing the OTA update. But I then need to reformat to mount it again which would erase any file that would have been written during the OTA update.

Any help would be most welcome!

suculent, 2018-02-28:
Wow, this is interesting. Unfortunately I need to delay solving this issue now, so before I dig in, some ideas. 

- Maybe, the download connection is not properly closed, because API does not send Connection: close. I’d have to look at that on the thinx-device-api side.

- Maybe, the downloaded firmware file is not closed/saved on the library side, but anyway: why does writing a file render spiffs useless? How big is it?

- Do you use SPIFFS or EEPROM to save the config? The THiNX library does format spiffs itself if it cannot SPIFFS.begin() for that purpose. 

This library does not handle spiffs, just expects it is formatted. 

Odesláno z iPhonu

gauix4, 2020-05-05:
impossible for me to update also it tells me ESP32 HTTP_UPDATE_FAILD Error (-100): To less space
I formatted the Flash and I sent a spiffs.bin file as light as possible a simple hello world in html so that it takes the least space possible but it always marks the same thing no matter the size of the file
my esp32 at 8M Flash it should largely contain my simple hello world
I did not find where it could come from the problem I tested initialization of the flash but nothing helps

suculent, 2020-05-11:
See TL;DR in README, this is already deprecated and implemented in Arduino Core.

closed 2020-05-11 by suculent