suculent/esp32-http-update #3: Q: still necessary?

marcelstoer, 2018-05-01:
Is this library in light of https://github.com/espressif/arduino-esp32/commit/febcda0095e9754787eeba3569543c953635da3e still necessary? Does it over any material benefit over what is in the Arduino core?

suculent, 2018-05-02:
> Is this library in light of espressif/arduino-esp32@febcda0 still necessary? Does it over any material benefit over what is in the Arduino core?

Hello, Marcel. 

I’ve done this library since this was not working before in Arduino core and I have some projects depending on it. 

Once those will be upgraded, library may be discontinued. 

I just have to find out how :o)

Odesláno z iPhonu

suculent, 2020-05-11:
See TL;DR in README, this is already deprecated and implemented in Arduino Core.

closed 2020-05-11 by suculent