suculent/esp32-http-update #6: This Library Work with HTTPS?

cruizg, 2019-04-01:
I have a problem

i need upload bin file with HTTPS, is posible when this library?

suculent, 2019-04-02:
You can use native Arduino Core update with HTTPS as of 2.5.0. 

The BearSSL there needs a bit different certificate format than this outdated example. 

M. 

Odesláno z iPhonu

cruizg, 2019-04-02:
Can you send me an example please because i am not understand? 
Thanks

suculent, 2019-04-02:
In Arduino IDE, see Examples > ES8266HTTPClient > BasicHttpsClient.

Docs for BearSSL are here: https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/bearssl-client-secure-class.html

cruizg, 2019-04-02:
that work with ESP32 ?

suculent, 2019-04-02:
Sorry, my mistake. For ESP32 it is in Arduino IDE Examples > HTTPClient > BasicHttpsClient. Once you manage working HTTPS connection, you're almost there and can try Examples > HTTPUpdate > httpUpdateSecure.

suculent, 2019-04-02:
See README (why do I write it if nobody reads it?)

This is just a quick port of ESP8266httpUpdate for ESP32. It is already deprecated, since Arduino Framework for ESP32 added this support officially. See: https://github.com/espressif/arduino-esp32/tree/master/libraries/Update

closed 2019-04-02 by suculent

cruizg, 2019-04-02:
very thans, 
I try that , the request HTTPS is correct but the request for update bin not work 
this is my code:

/**
   BasicHTTPSClient.ino

    Created on: 14.10.2018

*/

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <HTTPClient.h>
#include <HTTPUpdate.h>
#include <WiFiClientSecure.h>

// This is GandiStandardSSLCA2.pem, the root Certificate Authority that signed 
// the server certifcate for the demo server https://jigsaw.w3.org in this
// example. This certificate is valid until Sep 11 23:59:59 2024 GMT
const char* rootCACertificate = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIFnzCCBIegAwIBAgIQXp9hBX2JRo4BAAAAAAgf0zANBgkqhkiG9w0BAQsFADBC\n" \
"MQswCQYDVQQGEwJVUzEeMBwGA1UEChMVR29vZ2xlIFRydXN0IFNlcnZpY2VzMRMw\n" \
"EQYDVQQDEwpHVFMgQ0EgMU8xMB4XDTE4MTAzMTIxMjcyN1oXDTE5MTAzMDIxMjcy\n" \
"N1owaTELMAkGA1UEBhMCVVMxEzARBgNVBAgTCkNhbGlmb3JuaWExFjAUBgNVBAcT\n" \
"DU1vdW50YWluIFZpZXcxEzARBgNVBAoTCkdvb2dsZSBMTEMxGDAWBgNVBAMTD2Zp\n" \
"cmViYXNlYXBwLmNvbTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAMbd\n" \
"hbVoJUm9sW6ewBRmQ1qsBkrgySPietMhxGRNVLv9dDRRV/3rtS9uPEYjTQvj1jW0\n" \
"8aK8t3hYIJc3gVTmoKG9kGQ6/6JSt6avbn0M4rDE7wnNOrczHJtAPX7hWwIao5Wk\n" \
"Mhvy0kYFGdPkREejMaxvS7/m7m5Zk9Q13VV1PPqFd76nJX8t0NjsJzfQtdQqZY8l\n" \
"pbBWsQuuEnQdrsaeczZH0W/gZRtrrFns6HdQH8MLHS9sWh1dcy+hLeXREhxIqSCY\n" \
"ot7D2F9siknOtfCYgwQY9rKrfmXMa0P0RT3ngNU6h3hWsvKCJ9fkTn79OwrlFyec\n" \
"r5Y8B729Kaf1AvrM1m0CAwEAAaOCAmgwggJkMA4GA1UdDwEB/wQEAwIFoDATBgNV\n" \
"HSUEDDAKBggrBgEFBQcDATAMBgNVHRMBAf8EAjAAMB0GA1UdDgQWBBRSrNpbSi9r\n" \
"HQjRoCgeQ51wS3oVTDAfBgNVHSMEGDAWgBSY0fhuEOvPm+xgnxiQG6DrfQn9KzBk\n" \
"BggrBgEFBQcBAQRYMFYwJwYIKwYBBQUHMAGGG2h0dHA6Ly9vY3NwLnBraS5nb29n\n" \
"L2d0czFvMTArBggrBgEFBQcwAoYfaHR0cDovL3BraS5nb29nL2dzcjIvR1RTMU8x\n" \
"LmNydDAtBgNVHREEJjAkgg9maXJlYmFzZWFwcC5jb22CESouZmlyZWJhc2VhcHAu\n" \
"Y29tMCEGA1UdIAQaMBgwCAYGZ4EMAQICMAwGCisGAQQB1nkCBQMwLwYDVR0fBCgw\n" \
"JjAkoCKgIIYeaHR0cDovL2NybC5wa2kuZ29vZy9HVFMxTzEuY3JsMIIBBAYKKwYB\n" \
"BAHWeQIEAgSB9QSB8gDwAHcApLkJkLQYWBSHuxOizGdwCjw1mAT5G9+443fNDsgN\n" \
"3BAAAAFmzD0XDQAABAMASDBGAiEAy3h3yWHaawMTGuyx/5LQu8U8PHJ1Kxb/DFht\n" \
"AdMF9OYCIQCOUAAxJz35kjLcb75TlpthkU4nriXBe0cnY2mhsiHW8gB1AFYUBpov\n" \
"18Ls0/XhvUSyPsdGdrm8mRFcwO+UmFXWidDdAAABZsw9FwYAAAQDAEYwRAIgSu++\n" \
"IjmwwI4oLWDE0X+60cD6+43cXdf/kLQLbVWYfnMCIDSpg/jHtE1gN2nupwexjQdr\n" \
"B72Pdtx7ig5v1EdDQCJXMA0GCSqGSIb3DQEBCwUAA4IBAQB5D4ZtOsWVWtZ0pOMI\n" \
"/zktk4bimqsGCYvFtuj0BkBUotMf6uO8PAaZAUuPi5tLZX9rVnSpz0dEuQiZq52R\n" \
"QMI7ryacwwQYakeDv1PcNDDGwiL3DOqmWPkGs9q4O2KMGJeHGAgn6N71cSG1Csbe\n" \
"HrdP+C/QJSaU7Smgj/lmqXMGuFpvTITx3u8ZLqj461dIjddNO8kCWTEtlf78DYam\n" \
"R89HEe+e+hJKBmgCVZ99Xy6axOU+DfFnGpK3URzAuGIMG20xChlzE31364cNWCwh\n" \
"M8XHakyE3bcsbS4AwN57JIBzkXceEYHxWfteRykFV9wNmbFMn/KJsYXKZAc+dF+O\n" \
"4B55\n" \
"-----END CERTIFICATE-----\n";

// Not sure if WiFiClientSecure checks the validity date of the certificate. 
// Setting clock just to be sure...
void setClock() {
  configTime(0, 0, "pool.ntp.org", "time.nist.gov");

  Serial.print(F("Waiting for NTP time sync: "));
  time_t nowSecs = time(nullptr);
  while (nowSecs < 8 * 3600 * 2) {
    delay(500);
    Serial.print(F("."));
    yield();
    nowSecs = time(nullptr);
  }

  Serial.println();
  struct tm timeinfo;
  gmtime_r(&nowSecs, &timeinfo);
  Serial.print(F("Current time: "));
  Serial.print(asctime(&timeinfo));
}

WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("esp8266", "Admin123");

  // wait for WiFi connection
  Serial.print("Waiting for WiFi to connect...");
  while ((WiFiMulti.run() != WL_CONNECTED)) {
    Serial.print(".");
  }
  Serial.println(" connected");

 // setClock();  
}

void loop() {

  if ((WiFi.status() == WL_CONNECTED)) { //Check the current connection status

    
    
    HTTPClient http;

    http.begin("https://fir-cloud-8d70f.firebaseapp.com/", rootCACertificate); //Specify the URL and certificate
    int httpCode = http.GET();                                                  //Make the request

    if (httpCode > 0) { //Check for the returning code

        String payload = http.getString();
        Serial.println(httpCode);
        Serial.println(payload);
        WiFiClientSecure client;
    client.setCACert(rootCACertificate);
     t_httpUpdate_return ret = httpUpdate.update(client, "https://fir-cloud-8d70f.firebaseapp.com/bin/StartCounter.ino.bin/");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
      }

    else {
      Serial.println("Error on HTTP request");
    }

    http.end(); //Free the resources
  }

  delay(10000);
}

suculent, 2019-04-02:
Sorry, I’m not Arduino tech support. 

Odesláno z iPhonu