suculent/esp32-http-update #7: Code crashes at httpUpdate

thedim-witted, 2019-04-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
Describe your problem here
I am trying to integrate http OTA update feature in the ESP32 WROOM board. My code has been perfect uptil now. However, as soon as I add the line for the http, the code crashes at runtime right at the httpUpdate line. The error code shows it is causing some problem with the httpClient code. However, I am not able to debug it.
I was able to reproduce the error with the below sketch.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/**
   httpUpdate.ino

    Created on: 27.11.2015

*/

#include <Arduino.h>

#include <WiFiMulti.h>

#include <HTTPClient.h>
#include <ESP32httpUpdate.h>

#define FIRMWARE_UPDATE_VERSION "2001"

String updateURL = "XYZ.php";

WiFiMulti wifiMulti;
WiFiClientSecure client;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

//  for (uint8_t t = 4; t > 0; t--) {
//    Serial.printf("[SETUP] WAIT %d...\n", t);
//    Serial.flush();
//    delay(1000);
//  }

  WiFi.begin("xxxxxxxxxx", "xxxxxxxxxx");
}

void loop() {

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  // wait for WiFi connection
  if ((WiFi.status() == WL_CONNECTED)) {

    t_httpUpdate_return ret1 = ESPhttpUpdate.update(updateURL, FIRMWARE_UPDATE_VERSION);

    switch (ret1) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
  }
}
```

### Debug Messages:
```

PC: 0x400d3eb0: HTTPClient::connect() at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 959
EXCVADDR: 0x00000000

Decoding stack results
0x400d3eb0: HTTPClient::connect() at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 959
0x400d3f3d: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 512
0x400d3ffb: HTTPClient::GET() at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\libraries\HTTPClient\src\HTTPClient.cpp line 458
0x400d4ebb: ESP32HTTPUpdate::handleUpdate(HTTPClient&, String const&, bool) (C:\Program Files at x86)\Arduino\libraries\esp32-http-update-master\src\ESP32httpUpdate.cpp line 199
0x400d509b: ESP32HTTPUpdate::update(String const&, String const&) (C:\Program Files at x86)\Arduino\libraries\esp32-http-update-master\src\ESP32httpUpdate.cpp line 48
0x400d1def: loop() at C:\Users\Steiner\AppData\Local\Temp\arduino_modified_sketch_277684/httpUpdate.ino line 63
0x400d793d: loopTask(void*) at C:\Users\Steiner\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2-rc1\cores\esp32\main.cpp line 25
0x4008ef29: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

suculent, 2019-04-13:
Is the esp able to fetch the file alone using WifiClient or WifiClientSecure?

Do you have also Apache log or whatever you serve the PHP from? Are there any successful connection attempts by the device?

M. 

Odesláno z iPhonu

thedim-witted, 2019-04-13:
Hey sorry to bother but I found the issue to be very simple and embarrassing. The OTA was actually working. However, the new code it was flashing was corrupt and the core was crashing due to that. My bad. It was very difficult to find this issue though. Thanks anyway.

closed 2019-04-13 by thedim-witted

suculent, 2019-04-13:
Glad to know it still works.

Cheers,
M.

franzuu, 2019-05-26:
Hi! I see this issue is closed, but can you post a working code for this example? Also do you mind sharing the http part of the webserve? I cannot seem to find a working example with currentVersion checking.

suculent, 2019-05-26:
Sorry, but I don't have currently any free time time to manage those examples and documantation. This library should be already deprecated, everything is already covered vy Arduino Core for ESP32.

All the update-related code is always tested using THiNXLib (https://github.com/suculent/thinx-firmware-esp3 <https://github.com/suculent/thinx-firmware-esp3>2) against firmware update deployer on currently running THiNX instance (http://github.com/suculent/thinx-device-api <http://github.com/suculent/thinx-device-api>) at https://rtm.thinx.cloud <https://rtm.thinx.cloud/>. THiNX manages versions on its own and the Update procedure can be temporarily disabled, or deferred with user/administrator consent through second channel. New version bump is triggered using git tags and backend compares version of device against latest available custom-built binary before offering an update.

M..