suculent/esp32-http-update #8: progress number, % or bar

fotosettore, 2019-12-05:
hi
is it possibile to see the progress of the download during ota ?
many thanks

suculent, 2019-12-05:
Hello,

I'd guess you'd have to download the file on your own and then perform update from SPIFFS with the Core library. At least that's what I would do.

However to transfer 512kB over WiFi these days, I would just not do that at all. There's low chance anyone will be looking at the device that precise second.

M.

fotosettore, 2019-12-05:
hi
i explain better my needs.
the file will be downloaded from my private cloud.  Before i check if current version is new. If yes, update automatically starts using 
t_httpUpdate_return ret = ESPhttpUpdate.update( OTAImageURL );
no info about the dowload : only 20 second of panic until all restart :-)
at today's date all works fine but ... a percent info of download on the my project display should be a nice option !
peppe

suculent, 2019-12-05:
Indeed, the 

> ESPhttpUpdate.update( OTAImageURL ) 
has no progress indication functionality at all.

See http://www.iotsharing.com/2017/11/firmware-update-ota-via-esp-http-webserver.html

suculent, 2019-12-05:
That said, you don't need update progress indicator, but download progress indicator and then strike the update using file from spiffs, instead of updating from the network stream.

See https://techtutorialsx.com/2019/07/21/esp32-arduino-updating-firmware-from-the-spiffs-file-system/

fotosettore, 2019-12-05:
thanks a lot for your fast reply
sorry but user that needs update of my project is 2000 km far away from me and is not skilled to use arduino ide :-(
all should be done by himself without remote desktop or similar ... 
more ... no web page to show progress but a SIMPLY NUMBER during downloading ....
it is not possible ? ok ... it's a pity and I take note of this

suculent, 2019-12-05:
It is possible, but not with esp32-http-update - this library is deprecated in favour of Arduino Core >2.5.1 and its examples. 

It had its purpose before 2.5.0 but now the core update has been revamped anyway. 

Odesláno z iPhonu

fotosettore, 2019-12-05:
thanks.... i'll try it !

suculent, 2020-05-11:
See TL;DR in README, this is already deprecated and implemented in Arduino Core.

closed 2020-05-11 by suculent